JvanKatwijk/eti-stuff #1: Protection Level 4A not supported

andimik, 2017-03-06:
Seems you have fixed the 4A issue only in qt-dab. eit-stuff still don't support protection level 4A.

JvanKatwijk, 2017-03-07:
OK, I just checked the code and it seems that the protection handling is
identical to that in qt-dab, so it must be something different. I'll look
into it.
jan

basicmaster, 2017-03-08:
3B is affected, too (11C Munich).

andimik, 2017-03-09:
Stefan: What about Schwarzwaldradio on 5C Bundesmux? They are in 2B.

basicmaster, 2017-03-09:
Tuning to 5C works, but everytime I click on the ETI button, `eti-frontend` crashes, so I cannot check:
```
*** Error in `./eti-frontend': free(): invalid next size (normal): 0x00007f9bb400fd30 ***
```

JvanKatwijk, 2017-03-09:
Just for my understanding:
a. you run the eti-frontend
b. you select channel 5C,
there is a signal, you see the program names, and then
c. when you click on the button "eti waits" the system crashes

That is correct?

If sou can you make a recording of the channel 5C, so that I can try the
same run

best
jan

basicmaster, 2017-03-09:
To be precise, I do the following:
- start `./eti-frontend -O - | dablin_gtk`
- select 5C
- enable autogain
- click on "START"; after some seconds the list fills with the programs
- click on "eti waiting" and a second later, the program crashes

How can I make this recording (I guess you don't mean an ETI or FIC dump)?

JvanKatwijk, 2017-03-09:
With the qt-dab, you select device, start/select channel 5C, then press
"dump to raw file", you get a menu,
select a file/filename and then the input is just written to that file.

jan

basicmaster, 2017-03-09:
Here it is (about 20s):
https://basicmaster.de/test/5C_DR_Deutschland.sdr.zip

JvanKatwijk, 2017-03-09:
It does not synchronize,
jan

basicmaster, 2017-03-09:
Strange....I had to set the gain value to 420 (it was 165 or so) - despite the enabled autogain.
I uploaded again 20s (same link).

JvanKatwijk, 2017-03-10:
Are you sure it is the eti-front that crashes? I ran the combination
eti-frontend/etibackend with no problems and a lot of sound.
But be sure that you do not press the "eti waits" button before you have
full ensemble description.
jan

basicmaster, 2017-03-10:
Yes; with `eti-frontend -O test.eti` does not make a difference. Using the .sdr dump I sent you it crashes as well.

This is the stack trace from `gdb`:
```
#0  0x00007ffff7a4bc37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff7a4f028 in __GI_abort () at abort.c:89
#2  0x00007ffff7a882a4 in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff7b966b0 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff7a9455e in malloc_printerr (ptr=<optimized out>, str=0x7ffff7b96828 "free(): invalid next size (normal)", action=1) at malloc.c:4996
#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:3840
#5  0x0000000000420c36 in eep_deconvolve::~eep_deconvolve() [clone .4356] ()
#6  0x0000000000423ef9 in etiGenerator::process_CIF(short*, unsigned char*, int) [clone .2380] ()
#7  0x0000000000423b27 in etiGenerator::run() ()
#8  0x00007ffff5d3f32f in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#9  0x00007ffff451c184 in start_thread (arg=0x7fffd3fff700) at pthread_create.c:312
#10 0x00007ffff7b0f37d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

JvanKatwijk, 2017-03-10:
That is interesting, here it runs flawless. From your message it seems it
crashes in ~eep_deconvolve. Can you compile with a -g flag on and then send
the stack dump, then it will tell the linenumber

Strange
jan

JvanKatwijk, 2017-03-10:
What version are you running? eep_deconvolve was ages ago replaced by
eep_protection

jan

basicmaster, 2017-03-10:
> What version are you running?

OK, this explains the issues. I used the binary from the folder `/eti-frontend/linux-bin` (which was there in January). Apparently the location has changed to now `/linux-bin`. With the latter version, the levels e.g. 3B and 2B work for me (regarding 4A I don't know; I don't receive a mux with it).

andimik, 2017-03-20:
4A works now :-) Thanks for fixing.

closed 2017-03-20 by andimik