JvanKatwijk/eti-stuff #8: eti-frontend Compile Error on RPi3

MkMunich, 2017-04-04:
Hi Jan,
there is an compilation error:
pi@raspberrypi:~/eti-stuff/eti-frontend/build $ make
[  2%] Building C object CMakeFiles/eti-frontend.dir/src/backend/spiral-code/spiral-sse.c.o
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C
/home/pi/eti-stuff/eti-frontend/src/backend/spiral-code/spiral-sse.c:23:23: fatal error: emmintrin.h: Datei oder Verzeichnis nicht gefunden
 #include <emmintrin.h>
                       ^
compilation terminated.
CMakeFiles/eti-frontend.dir/build.make:106: recipe for target 'CMakeFiles/eti-frontend.dir/src/backend/spiral-code/spiral-sse.c.o' failed
make[2]: *** [CMakeFiles/eti-frontend.dir/src/backend/spiral-code/spiral-sse.c.o] Error 1

The following did not help to insert into eti-frontend.pro file:
CONFIG += NO_SSE_SUPPORT

JvanKatwijk, 2017-04-04:
Yes, it contained code from centuries ago, I just removed the "sse"
paragraph and it compiled fine with me

MkMunich, 2017-04-06:
Hi Jan, I tried compilation again using the updates. Now it is compiling DAB_STICK but linking issues an error:
[100%] Linking CXX executable eti-frontend
/usr/bin/ld: cannot find -lQt5::Network
collect2: error: ld returned 1 exit status
CMakeFiles/eti-frontend.dir/build.make:950: recipe for target 'eti-frontend' failed
make[2]: *** [eti-frontend] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/eti-frontend.dir/all' failed
make[1]: *** [CMakeFiles/eti-frontend.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

I've to admit, that I ignored the cmake warning:
  Target "eti-frontend" links to target "Qt5::Network" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
  This warning is for project developers.  Use -Wno-dev to suppress it.

Btw: I sudo apt-get installed qt5, but so far I can see from the following command + output, only libqt5network5 is available in my qt5 installation on my RPi 3 Raspian Jessie:
    apt-cache search qt5 | grep network
libqt5network5 - Qt 5 network module

Any ideas?

JvanKatwijk, 2017-04-06:
I am afraid the CMake files heavily depend on Qt5, Can't you use the
qmake/make route instead? On my rpi2 I always use the qmake-qt4 / make
route without problems

JvanKatwijk, 2017-04-06:
Completely different question:

What - as user - do you want  on a command line interface for DAB. I cannot
garantee that everything will/can be implemented but the underlying
question is that what I really would like to have is a separate library and
various interfaces - both with GUI and command line like, since then there
can be a clear distinction between the DAB functionality and the
presentation  (I am jst a programmer, personally I do not care at all about
GUI's etc)

best
jan

MkMunich, 2017-04-06:
As I'm only building DABSTICK, it helped to just comment out the reference "Qt5::Network" in section target_link_libraries(...) in file CMakeLists.txt. I needed the same in eti-backend. then, both applications linked.

From this you wrote above I would guess, that you're using the older qt4. It is a bit misleading in my opinion. May be an option to delete qt5, from the build settings, then?

MkMunich, 2017-04-06:
Sorry, I forgot to answer your question in your last post: Actually, I do not know yet and wanted to ask you, whether better to continue/use here or dab-cmdline. What I appreciate is, that GUI and background processing is separated. This has several advantages: GUI could be replaced e.g. by a kodi application as I'm looking at. An other point is the different qt versions. If separated it is easier to update and the background is not effected.

What I need to solve for RasPiCar is ~ the following:
a. How to integrate dab... into Kodi from RasPiCar.
b. 1. RasPiCar will need a list of stations to chose during driving.
b. 2. Because the station list will be not be static, when driving, RasPiCar would need a 'scan' functionality. Nice to have would be, when a station is lost, then it should try to find the same station somewhere else (means an other frequence like FM car radios are offering)
c. DAB+ means different streams: audio and info + gif, etc., which must be provided in Kodi. Audio may be easy, as the stream might be sent to the ALSA mixer device as drain.
d. I have no idea about the other streams like program information + gif, etc., yet.

Well, a lot fun :-)

JvanKatwijk, 2017-04-06:
I got several request to have something for Kodi, but I am not familiar
with it, but with some help I would like to make the library approach
suitable - probably with some interface layer - for Kodi.

starting slowly:
Would you need to be able to dynamically switch between stations? The
current implementation is more or less focused to receiving one station
(the original approach was to implement - on request - a command line
implementation, when doing so I figured it would be a good idea to
implement some kind of library)
If so then I will adapt the API and the library to handle "search a
different program in the current list"

Btw, the list of programs is handed over to whatever the GUI is by the
library when found.

The dynamic parts need some scanning function, but I an not sure yet what
it should do exactly
best
jan

MkMunich, 2017-04-06:
OK. Good idea to go step by step.

First to be sure, that I got it: you are suggesting to take dab-cmdline/dab-library as starting point?
(If so, then we might open an issue in dab-cmdline rather than here in eti-stuff?)

Yes, one of the first steps should be to enable Kodi to choose a radio station. For this, a list of radio stations should be able to be created. Let's call this function 'Scan'. In principle, 'Scan' is located in a logical layer in between. From the history "qt-dab", perhaps an idea to leave Scan in the core layer, but separate it to group the functionality of radio station handling. So this layer could be called 'StationHandler'. The purpose of StationHandler would be to maintain a list of radio stations. Scan would fill the list. May be, there are two Scan options: 1. Clear the list in advance or 2. merge the radio stations. Alternatively Scan could always merge and there is a separate 'Clear'. Merging sounds a bit more work, but is more user friendly as the radio list order could remain the same. A detail would be to define, that radio stations, which are no longer receivable (because the car moved out of the coverage) are left in the radio station list marked as currently not receivable, so in case the car is driving back, the station will be present again and the flag is reset. To me it looks, that one frequency is providing a set of stations at once. So in case the car is driving out of a transmission area the actual playing radio station will stop to provide an audio stream. This could be used to detect it. I'm guessing, that all radio stations of this frequency are then 'lost'. May be an idea for future, to start a scan (merge) in order to find the same radio station on an other frequency.
Main functions called from Kodi would be:
- issue Scan and Clear commands
- Set the actual radio station to start playing. (May be an option to implement a Start and Stop playing).
- Set callbacks to receive the information streams and gifs, etc. This callback functions are of course a layer of Kodi. As soon the callback function is registered, Kodi should be able to process (display) the information.
- Retreive the (actual) list of radio stations and its flags, whether currently available.
- Change order (may be not the first function to implement ;-)

I'll stop for now to give you a chance to think and comment my ideas...

andimik, 2018-01-29:
Is this issue still open? @MkMunich

closed 2018-03-01 by JvanKatwijk