JvanKatwijk/eti-stuff #10: seg fault after pressing eti writing

andimik, 2017-04-18:
When I write an ETI and want to stop it there is a seg fault

JvanKatwijk, 2017-04-19:
but the file is written? there are some issues with Qt5 and program
termination

andimik, 2017-04-19:
Yes, file is written

JvanKatwijk, 2017-04-19:
OK

It would be really helpful if you can send a link to a file where there is
a noticeable difference in the behaviour using the januari version  vs the
recent version of the eti-frontend. Running here - with an airspy as input
device - does not show mch difference (note that the january version
contains the error with handling protection level 4, and incomplete
versions of pad handling, so it need some updating before promoting it to
the "current" version).

Again, I am not very satisfied with the eti-frontend as it is, it needs a
redesign on its input. If we would accept that running the frontend on the
raspberry 2 is not an option, internal operation could be simplified a lot

JvanKatwijk, 2017-04-19:
I just had 2.5 hours music using the eti-commandline + eti-backend
combination working fine. I suggest further development of the commandline
handler - i.e. adding file input, better defining the parameters - and
withdrawing the Qt based version. The latter needs too much work before all
issues

andimik, 2017-04-19:
I cannot test eti-cmdline because it seems not to be configured for file input.

What happens when you pipe eti-frontend to eti-backend?

`eti-frontend -O - | eti-backend -s -
`

Newer versions (later than January) have lots of dropouts, but it could be just a break between the audio frames. Because when I record it through the "audio" button, the sound is ok and the audio is playing normally.

See https://github.com/JvanKatwijk/qt-dab/issues/22

The soundcard output is indeed that

[image]

I would suggest to go back to this January version and fix the 4A issue, the file extension issue, and the ETI file handling issue for Windows.

andimik, 2017-04-20:
Thanks for your PM (I also have replied personally as well, ;-) ).

The eti-cmdline now looks pretty good and is very simple to install (not compared to that one on github). 

My aim is just to record DAB muxes wherever I am (on vacation, etc) and replay them later and to minimize filesizes (-> eti is the best way).

JvanKatwijk, 2017-04-20:
I assume(d) your preference was to record 8 bit streams. I intend to add
the 16 bit sdr files as well (and add some testing stuff)

One of the advantages of the command line version is that all parameters
are fixed at program startup time.
If you have special issues on this eti-cmdline thing, pls let me know

andimik, 2017-05-23:
Closed as not more necessary due to the new eti-cmdline implementations

closed 2017-05-23 by andimik