JvanKatwijk/eti-stuff #16: eti-cmdline-rawfiles and *.sdr support

andimik, 2017-05-20:
As qt-dab saves dumps as *.sdr (s16le resp s16be), why not add support for it in eti-cmdline?

JvanKatwijk, 2017-05-21:
I'll look into it. The obvious next question then will be: "why can't you
read in ".sdr" files"

JvanKatwijk, 2017-05-21:
You could give eti-cmdline a try, I added both a dumping facility (add
-DDUMPING=ON to the cmake command) and a WAV file option
for reading ".sdr" files

I ran a few tests and it seems to work

2017-05-21 11:10 GMT+02:00 jan van katwijk <j.vankatwijk@gmail.com>:

> I'll look into it. The obvious next question then will be: "why can't you
> read in ".sdr" files"
>
>
>
>> As qt-dab saves dumps as *.sdr (s16le resp s16be), why not add support
>> for it in eti-cmdline?
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

andimik, 2017-05-21:
No. Seg fault :-(

> andreas@andiE535:~$ eti-cmdline-wavfiles -F 10b_ras_kurz.sdr
> Period = 16000
> Bus-Zugriffsfehler (Speicherabzug geschrieben)

The sdr was generated with Airspy.

Installing:

> andreas@andiE535:  /home/andreas/eti-stuff/eti-cmdline/build_wavfiles$ cmake .. -DWAVFILES=ON
> -- The C compiler identification is GNU 5.4.0
> -- The CXX compiler identification is GNU 5.4.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Build type not specified: defaulting to release.
> -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
> -- Checking for module 'fftw3f >= 3.0'
> --   Found fftw3f , version 3.3.4
> -- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
> -- Found LibSndFile: /usr/lib/x86_64-linux-gnu/libsndfile.so
> -- Found LibSampleRate: /usr/lib/x86_64-linux-gnu/libsndfile.so
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/andreas/eti-stuff/eti-cmdline/build_wavfiles
> andreas@andiE535:~/eti-stuff/eti-cmdline/build_wavfiles$ 
> 
then `make` and `sudo make install`

JvanKatwijk, 2017-05-22:
Strange, works like  charm here

[jan@fedora-hp ~]$ eti-cmdline-wavfiles -F Downloads/10d_04022017a.sdr >
/dev/null
Period = 16000
ensemble MULTIPLEX R1     detected
program ( 1)t Slo 1 Prvi           9201 is in the list
program ( 2)t Slo 3 ARS            9203 is in the list
program ( 3)t Slo 2 Val 202        9202 is in the list
program ( 4)t Slo Int Radio SI     933A is in the list
program ( 5)t Radio Ognjisce       942F is in the list
program ( 6)t Radio Center Slo     9414 is in the list
program ( 7)t Radio 1 DAB          9442 is in the list
program ( 8)t Radio Antena         940B is in the list
program ( 9)t Radio 2              9459 is in the list
program (10)t Radio BOB            945B is in the list
program (11)t Rock Radio Slo       9516 is in the list
program (12)t Glasbeni Radio       9462 is in the list
program (13)t A-Radio              9436 is in the list
there might be a DAB signal here
still at most 10seconds to wait
yes, here we go
Handling ensemble MULTIPLEX R1    until you quit
^C65        estimated snr:  0, fibquality   0

The thing I have to address now is to gracefully terminate after reaching
eof with both raw and wav files

andimik, 2017-05-23:
Thanks for the new version, works now. :-) 

I have a very weak recording from a signal around 200 km away, sdr is the only format where I have audio. As soon as I convert it into 8 bit raw, the SNR is too poor for any tool.

That's why this new version is like a present for me.

closed 2017-05-23 by andimik

JvanKatwijk, 2017-05-23:
good to hear