JvanKatwijk/eti-stuff #17: eti-cmdline: how to get rtl_tcp working as input device

chrwick, 2017-09-19:
Hi!
Could you please give me some advice on how to get the rtl_tcp input working?

JvanKatwijk, 2017-09-19:
Can you be more specific? the input port is 1234
I assume you have the rtl_tcp server working

If you touch the connect button it will show a line where you can enter the
server address

andimik, 2017-09-19:
Seems he is talking about the command line version. There is currently no support for it, only for the five mentioned under https://github.com/JvanKatwijk/eti-stuff/blob/master/eti-cmdline/README.md

JvanKatwijk, 2017-09-20:
> JvanKatwijk/eti-stuff/blob/master/eti-cmdline/README.md

Yes sorry I did not notice that. As Andreas notes, rtl_tcp is not standard
implemented in the cmdline version
but it is not hard to do, so give it a try. I do not have time right now.

chrwick, 2017-09-22:
Ok. Thanks.

Meanwhile I have found a way to get it working via rawfiles and a pipe.

I have setup the RTL_TCP server to the desired frequency and gain. 
Then I have used netcat to connect to the RTL_TCP server port and pipe the stdout to a fifo created with mkfifo.

This fifo file is then used for eti-cmdline-rawfiles:
./eti-cmdline-rawfiles -F ./test -O -

The stdout of eti-cmdline-rawfiles is then piped to final tool which handles the ETI stream.
(In my case ni2http from [here ](https://github.com/piratfm/eti-tools))
However, I am now experiencing [this weird issue](https://github.com/piratfm/eti-tools/issues/10).

Closing this issue.

andimik, 2017-09-22:
Your idea does not work. Please forget all normal players. For ETI Frontend you should pipe it to dablin_gtk or dablin.

chrwick, 2017-09-22:
Why should it not work?
It IS already working with ni2http as it streams the pure AAC stream to a file or to an icecast2 server.
It works with VLC and my Sonos speakers listening to the icecast2 stream.

However, the only problem left is that it plays slightly to slow so as described in the other issue. 
The question for me is: whose fault is it? eti-cmdline-rawfiles or ni2http

What could be the reason for this?

andimik, 2017-09-22:
We are talking about DAB, the fault is the spec ;-) 

AAC for files or streams is not the same as AAC in DAB.

JvanKatwijk, 2017-09-23:
I had some spare time and added a version of rtl_tcp to the eti-cmdline
stuff. Unfortunately, while I can see that there
is a connection with an rtl_tcp server, I cannot see the full operation,
since where I am right now, there is no decent DAB signal

You could give it a try or wait until monday.
Relevant parameters
-H hostname
-I  port
-G the gain is NOT in the range 0 .. 100 as with other devices, but in
tenth of db

best
jan

chrwick, 2017-09-25:
Thanks!
Seems to be working for. It outputs an ETI stream to which I can listen with ni2http. 
My other [issue](https://github.com/piratfm/eti-tools/issues/10) is still there.

chrwick, 2017-09-28:
Tried RTL_TCP with dablin and I get:

```
ETIPlayer: ignored ETI frame with ERR = 0x55
ETIPlayer: ignored ETI frame with ERR = 0x55
ETIPlayer: ignored ETI frame with ERR = 0x55
ETIPlayer: ignored ETI frame with ERR = 0x55
```

JvanKatwijk, 2017-09-29:
My mistake. It happened that the rtl_tcp handler wrote some stuff to
stdout. When piping the output, the erroneously written text
preceded the data in the eti file, and syncing was lost.

I tested it writing output to a file, that - obviously - went well, so it
took a while before I found out what was wrong

Corrected

closed 2017-10-02 by chrwick