JvanKatwijk/eti-stuff #18: eti-cmdline: phaseReference missing

basicmaster, 2018-01-28:
The current version of `eti-cmdline` cannot be compiled, as the definition of `phaseReference` is missing. When I go back to the previous commit, compilation works.

Used `cmake` call:

    cmake .. -DRTLSDR=ON

Output:
```
[...]/eti-stuff/eti-cmdline/src/ofdm/ofdm-processor.cpp: In constructor ‘ofdmProcessor::ofdmProcessor(deviceHandler*, SNDFILE*, uint8_t, void*, syncsignal_t, snrsignal_t, etiGenerator*, int16_t, uint8_t)’:
/home/stefan/Schreibtisch/DL-Source/eti-stuff/eti-cmdline/src/ofdm/ofdm-processor.cpp:65:66: error: no matching function for call to ‘phaseReference::phaseReference(dabParams*, int16_t&)’
                                                         threshold) {
                                                                  ^
```

JvanKatwijk, 2018-01-28:
should be fixed by now.

basicmaster, 2018-01-29:
Now it works; thanks!

closed 2018-01-29 by basicmaster