JvanKatwijk/eti-stuff #19: eti-cmdline: add option to disable status

basicmaster, 2018-01-29:
It would be great if a command line parameter could be added, which disables the permanent status output like

    125     	estimated snr: 11, fibquality 100

The reason is that when the ETI output is forwarded to another tool (e.g. DABlin), the `stderr` outputs might be mixed, especially when due to bad reception RS coding errors are found, fixed and output to the user.

JvanKatwijk, 2018-01-29:
added a -S switch for being silent

closed 2018-01-29 by basicmaster

andimik, 2018-02-10:
This does not work for rawfiles and for wavfiles:

`eti-cmdline-wavfiles -F foo.sdr -S  > foo.eti`
or
`eti-cmdline-rawfiles -F foo.raw -S  > foo.eti`

will both show only the help window.

JvanKatwijk, 2018-02-10:
Good point. The handling of the -S option (and the -h option) happened to
be neclosed in a part that was enclosed in the wrong part of an ifdef ..
endif construct