JvanKatwijk/eti-stuff #27: Update info of RTL-TCP input

lars18th, 2018-02-16:
Hi,

In the Homepage (`README.md`) isn't indicated the parameter for configure&compile the RTL-TCP module:

> where XXX refers to the input device being supported, one of (RTLSDR, SDRPLAY, AIRSPY, RAWFILES, WAVFILES)

You need to complete to the list: AIRSPY, SDRPLAY, RTLSDR, RAWFILES, WAVFILES, **RTL_TCP**.
As indicated in the `CMakeLists.txt` file: 
https://github.com/JvanKatwijk/eti-stuff/blob/master/eti-cmdline/CMakeLists.txt#L60

Regards.

lars18th, 2018-02-16:
Hi,

Futhermore, the `eti-cmdline-*` binary needs to print in the help the parameters of the module.

For example, for RTL_TCP these two parameters exists: `./eti-cmdline-rtl_tcp -H <ip> -I <port>`
But the program prints:

```
./eti-cmdline-rtl_tcp: invalid option -- '?'
 eti-cmdline-xxx options are

   -D number   time (in seconds) to look for a DAB ensemble
   -B Band     select DAB Band (default: BAND_III, or L_BAND)
   -P number   PPM correction
   -C channel  DAB channel to be used (5A ... 13F resp. LA ... LP)
   -G Gain     gain for device (range 1 .. 100)
   -Q          autogain for device (not all tuners support it!)
   -F filename load samples from file
   -E          only for files: continue after EOF (replay file)
   -O filename write output into a file (instead of stdout)
   -S          do not display quality messages while running
   -R filename (if configured) dump to an *.sdr file

   -h          show options and quit
```

I hope you can improve it! :smile:

andimik, 2018-02-16:
You can contribute by changing the file and send a Pull request

JvanKatwijk, 2018-02-16:
well, any implemented piece of software  has "undocumented features", this
happens to be one of them.
I am not in favour of extending the READme and the help file to tell there
features since usually the only reason they are there
is that these were created to test something and their functionality might
not be complete

lars18th, 2018-02-16:
Hi @JvanKatwijk ,

> is that these were created to test something and their functionality might
not be complete

So, you don't have planed to full support RTL_TCP? Why not?

JvanKatwijk, 2018-02-16:
why should I?

lars18th, 2018-02-16:
> why should I?

Because it just works and it's already implemented. No?

JvanKatwijk, 2018-02-16:
No way, but if you feel it has to be changed, submit a pull request

closed 2018-03-06 by JvanKatwijk

lars18th, 2018-03-06:
Hi @JvanKatwijk ,

> if you feel it has to be changed, submit a pull request

Following your suggestion I created a PR #35 for it.