JvanKatwijk/eti-stuff #28: [Suggestion] Windows version for the backend

lars18th, 2018-02-16:
Hi,

I put this here only as an idea: create a Windows release for the backend.

It can be possible using Qt 5.0 and MinGW.

Regards.

andimik, 2018-02-16:
There was a Windows version on https://www.sdr-j.tk/index.html but the latest zip-file does not contain it any more.

JvanKatwijk, 2018-02-16:
There is no support for the eti-backend anymore since there is an excellent
alternative with the dablin stuff.  I did not look at it for a long time

lars18th, 2018-02-16:
> There is no support for the eti-backend anymore since there is an excellent
alternative with the dablin stuff

I agree! DABlin, qt-dab and javaDAB are all very good.
However, with the exception of DABlin that isn't portable (Linux only) the other two lack support for ETI input from "eti-cmdline". I hope soon one of them can handle this input.

Regards!

andimik, 2018-02-20:
I just tried eti backend under Windows (version from April 2017). Believe me, it's better not to have it. So even please don't ask for it.

lars18th, 2018-02-20:
Hi @andimik ,

> I just tried eti backend under Windows. Believe me, it's better not to have it. 
> So even please don't ask for it.

OK. Then my objective now it's get that other tools (aka DAB players) will support natively ETI input.

closed 2018-02-20 by lars18th

JvanKatwijk, 2018-02-20:
Well, the later backends run better (just ran a test under W10 with file
input, works fine), but spending time/work on the ETI backend is not a
priority for me,
To make it into a real back-end, attention should be paid to erroneous
input, and the back-end code should be made inline with that of
the qt-dab program. Lots of work, and why???

lars18th, 2018-02-20:
Hi @JvanKatwijk ,

> Lots of work, and why???

I know why not develop a lot more the ETI backed. However, it's good to work on obtain a _decoupled_ DAB demodulator (frontend) from the DAB decoder (backend). As this will simplifies the work done by the decoder. And in this scenario the ETI bitstream can be a good interchange format.

So, the only problem now it's to expand the current DAB players for supporting the ETI input.

Thank you for all your good work! :smile: