JvanKatwijk/eti-stuff #29: Add option for time sync timeout

basicmaster, 2018-02-16:
The `waitingTime` can be set by `-D` to cope with bad reception of a DAB channel. Just before that, some kind of time sync happens (using `timesyncSet` and `timeSynced`) which quite fast decides whether a DAB signal is found or out.

It would be great if also the waiting time for this time sync could be set/increased via parameter, as this would allow to adjust the antenna while waiting for a sync. Otherwise, `dab2eti` quite fast exits with "There does not seem to be a DAB signal here" as no signal was found so far.

JvanKatwijk, 2018-02-16:
Syncing is in 2 steps. If time synchronization can not be achieved within a
reasonable amount of time it is assumed that there is no signal, so further
trying does not
make sense. Step 2 is waiting until we can actually interpret the data.

However, "making sense" or not is an interpretation done by the main
program. That just gets a signal through a call back function. If no one
reacts on that
signal, then everything just goes on.
If the main program does not receive within 5 seconds a signal that a time
sync could be achieved, it now decides to exit.
I will add a "-d xxx" parameter with which this maximum waiting time can be
set

JvanKatwijk, 2018-02-16:
Added a '-d' option as mentioned. In the setup now the -D delay starts
after the -d delay,

2018-02-16 20:29 GMT+01:00 jan van katwijk <j.vankatwijk@gmail.com>:

> Syncing is in 2 steps. If time synchronization can not be achieved within
> a reasonable amount of time it is assumed that there is no signal, so
> further trying does not
> make sense. Step 2 is waiting until we can actually interpret the data.
>
> However, "making sense" or not is an interpretation done by the main
> program. That just gets a signal through a call back function. If no one
> reacts on that
> signal, then everything just goes on.
> If the main program does not receive within 5 seconds a signal that a time
> sync could be achieved, it now decides to exit.
> I will add a "-d xxx" parameter with which this maximum waiting time can
> be set
>
>
>
>
>
>> The waitingTime can be set by -D to cope with bad reception of a DAB
>> channel. Just before that, some kind of time sync happens (using
>> timesyncSet and timeSynced) which quite fast decides whether a DAB
>> signal is found or out.
>>
>> It would be great if also the waiting time for this time sync could be
>> set/increased via parameter, as this would allow to adjust the antenna
>> while waiting for a sync. Otherwise, dab2eti quite fast exits with
>> "There does not seem to be a DAB signal here" as no signal was found so far.
>>
>>
>
>
>
> Jan van Katwijk
>
>
> +31 (0)15 3698980 <+31%2015%20369%208980>
> +31 (0) 628260355 <+31%206%2028260355>

closed 2018-02-18 by basicmaster