JvanKatwijk/eti-stuff #30: Paths with space don't work

andimik, 2018-02-22:
I wanted to use a file on my DVD which has a space in its filename:

> $ eti-cmdline-rawfiles -F /media/andreas/DAB_europa/9D_SWR\ BW\ N_2018-01-06_112554.raw > dab/9D_SWR.eti

But it seems escaping did not work in the program:

> could not open /media/andreas/DAB_europa/9D_SWR BW N_2018-01-06_112554.raw

Then I copied it to /tmp and renamed it and it worked

> $ eti-cmdline-rawfiles -F /tmp/9D.raw > dab/9D_SWR.eti
> Period = 8000
> there might be a DAB signal here

andimik, 2018-02-22:
BTW: dab-cmdline works with spaces ...

JvanKatwijk, 2018-02-22:
What would happen is you changed the filename in /tmp to contain a space. I
just copied a file for abc\ def.sdr and processing was no problem,
so it might be related to using the DVD

andimik, 2018-02-22:
Bingo! Eti-stuff still had this r+b issue in the xxx-handler.cpp

andimik, 2018-02-22:
Sent a PR

closed 2018-02-22 by andimik

andimik, 2018-02-23:
Fixed in  f39cfe2

JvanKatwijk, 2018-02-23:
See, you are becoming a programmer!!!