JvanKatwijk/eti-stuff #39: facing audio glitch

preenu, 2019-01-15:


JvanKatwijk, 2019-01-15:
can you be slightly more specific

preenu, 2019-01-16:
I am facing some glitches when eti file is played. I have attached the log below:

SDLOutput: using SDL version '2.0.2'
ETIPlayer: playing sub-channel 0 (DAB+)
ETISource: reading from 'dabplustest1.ETI'
FICDecoder: SubChId  0: start   0 CUs, size  96 CUs, PL EEP 3-A = 128 kBit/s
FICDecoder: SId 0xE001: audio service (SubChId  0, DAB+, primary)
FICDecoder: SId 0xE001, SCIdS  0: MSC service component (SubChId  0)
FICDecoder: EId 0xE000: ensemble label 'REDWOOD'
FICDecoder: SId 0xE001: programme service label 'REDWOOD SRV00'
(0+) SuperframeFilter: Superframe sync started...
(5+) (0+) (0+) SuperframeFilter: Superframe sync succeeded after 3 frame(s)
ETIPlayer: format: HE-AAC v2, 48 kHz Stereo @ 128 kBit/s
AACDecoder: using decoder 'FAAD2'
SDLOutput: using audio buffer of 192000 bytes
SDLOutput: audio opened; driver name: pulseaudio, freq: 48000, channels: 2, size: 9216, samples: 1152, silence: 0x00, output: 32bit float
(AU #2) (0+) (AU #2) (0+) (AU #2) (0+) (AU #2) (0+) (AU #2) (0+) (AU #2) SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (6144 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (24576 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...

preenu, 2019-01-16:
> And do the same errors occur in dablin_gtk?
>
> Because I also have them and fear they are coming not from the DAB tools,
> but from sdl.

Yes, i am facing the same problem in dablin_gtk.
Thanks & regards
Preenu Ann P

basicmaster, 2019-01-18:
It seems that the posted log only affects DABlin, but not eti-cmdline. As the issue creator is no longer registered on GitHub, I propose to close the issue.

If needed, a new issue can be opened for [DABlin itself](https://github.com/Opendigitalradio/dablin/issues).

closed 2019-01-18 by JvanKatwijk

andimik, 2019-01-18:
That's why he is called "ghost" 😊