JvanKatwijk/eti-stuff #40: eti-cmdline: not able to generate DAB+ ETI file

aleena57, 2019-01-21:
I could generate DAB file but am unable to generate DAB+ file. While playing the audio through dablin_gtk, audio is not clear.

andimik, 2019-01-21:
Could you please upload this file and post all messages?

What source do you use? How is the reception?

aleena57, 2019-01-21:
ETIPlayer: playing sub-channel 0 (DAB+)
ETISource: reading from 'dabplustest2.ETI'
FICDecoder: SubChId  0: start   0 CUs, size  96 CUs, PL EEP 3-A = 128 kBit/s
FICDecoder: SId 0xE001: audio service (SubChId  0, DAB+, primary)
FICDecoder: SId 0xE001, SCIdS  0: MSC service component (SubChId  0)
FICDecoder: EId 0xE000: ensemble label 'REDWOOD'
FICDecoder: SId 0xE001: programme service label 'REDWOOD SRV00'
(0+) SuperframeFilter: Superframe sync started...
(0+) SuperframeFilter: Superframe sync succeeded after 1 frame(s)
ETIPlayer: format: HE-AAC v2, 48 kHz Stereo @ 128 kBit/s
AACDecoder: using decoder 'FAAD2'
SDLOutput: using audio buffer of 192000 bytes
SDLOutput: audio opened; driver name: pulseaudio, freq: 48000, channels: 2, size: 9216, samples: 1152, silence: 0x00, output: 32bit float
(AU #2) (5+) (AU #2) (0+) (AU #2) (0+) (AU #2) (0+) (AU #2) SDLOutput: silence started...
(5+) (AU #1) (AU #2) SDLOutput: silence ended (9216 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (30720 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (15360 bytes)
SDLOutput: silence started...
(0+) (AU #2) SDLOutput: silence ended (6144 bytes)
SDLOutput: silence started...

aleena57, 2019-01-21:
source is REDWOOD streamer.

aleena57, 2019-01-21:
> Ok, then the file should be error-free of course.

Then what would be the issue?

basicmaster, 2019-01-21:
This is difficult to say without having the actual ETI file. I tend to assume that the file content is not 100% correct for some reason. Is it [this device](http://www.redwoodcomm.com/product/01.php?cate_1=36&cate_2=42) with which you created the mux?

And with which hardward/software did you get the signal to DABlin? This cannot be seen in the posted logs, as you just play the file `dabplustest2.ETI` here.

Could you upload the ETI file somewhere? A minute or so would be enough, if it is too large.

BTW did you also open JvanKatwijk/eti-stuff#39 some days ago?

JvanKatwijk, 2019-01-21:
> Is it this device with which you created the mux? Yes
> And with which hardward/software did you get the signal to DABlin? dabstick

we can see that the eti file does not give the good results, but can we get
a recording of the input?
with an input file I can do some checking and see where things go wrong

aleena57, 2019-01-22:
I don't have recording of input. I only have the eti file.

andimik, 2019-01-22:
You wrote you made the recording with a dabstick. So you can write the input file (raw or sdr file).

JvanKatwijk, 2019-01-22:
> :
> I ran a quick test and I noticed that the reed-solomon decoding for DAB+
> frames consistently give more errors
> than can be repaired.
> Are we sure that the incoming data is correct?

I ran some further tests and it shows that the data in the MSC is
corrupted, the data in the FIC is apparently not.
Looking at the et-cmdline sources (which I did not do for quite a while)
shows that handling the MSC is unaware
of the kind of service carried in the MSC, it handles "subchannels", so
DAB, DAB+ and data channels are all handled
the same.

So it is really necessary to get the incoming data. Until then I cannot be
certain that the incoming data is correct

basicmaster, 2019-01-22:
- as far as I understand https://github.com/Opendigitalradio/dablin/issues/21#issuecomment-455988549, DAB (MP2) works fine
- the first and the second AU of the three AUs of almost every Superframe is error free (all CRCs correct)...the third AU never though...such a pattern certainly cannot result from a lower layer
- the PAD of the error-free AUs is not compliant (the typical X-PAD len issue; use `dablin_gtk` instead of `dablin` to get the log messages)

So my current assumption is that the generator already outputs faulty DAB+. At least the PAD is affected...

andimik, 2019-01-22:
So let's wait until we get a rawfile

andimik, 2019-01-23:
You mean the sampling rate?

basicmaster, 2019-01-23:
@aleena57: While your issue is solved now, it would be really helpful if you could provide the raw dump anyway, so that we can investigate if there is an issue on our side. I don't think so, but I would like to be sure ;-)

JvanKatwijk, 2019-01-24:
It seems @aleena57 does not react anymore on the question to make the input available, so I suggest to close the issue. However, it seems not quite in line with the net-etiquette to not answer, since we all spent time addressing his/hers question. 
The question whether or not he/she also was responsible for issue #39 was also not answered.
Sad

closed 2019-01-24 by JvanKatwijk

basicmaster, 2019-01-24:
Yes, it's indeed annoying when people take, but don't give back....this would be the least...