JvanKatwijk/eti-stuff #41: Compiling eti-commandline in Ubuntu 18.04

Chris-v-d-West, 2019-03-21:
Hi Jan,

I am trying to compile eti-commandline.  I followed the README.md, using cmake .. -DRTLSDR=ON

I get a whole bunch of errors starting with

/eti-cmdline/main.cpp:360:2: error: #elif after #else

My C++ was never good and now rusty as well.  Never been good with these conditional directives.  On lines 300 and 313, why de you have #ifdef	HAVE_SDRPLAY and #if defined (HAVE_SDRPLAY)?

dab2eti does not work for me, is say PLL not locked, which is why I'm trying this.

We restarted DAB+ trial transmission in Johannesburg and Pretoria last week.  I'm using Qt-DAB on Windows at work and even the Windows installer on Ubuntu at home, with rtl_tcp.  Now I need something to get from RF to ETI, to use many ETI tools.

eti-backend compiled, just with warnings.  I haven't tried it, having no ETI feed!

Thanks a lot.

Chris

JvanKatwijk, 2019-03-21:
Thanks for the message. Somehow there was a mixup betweet versions that
should not have stored to the repository,
based on some previous comments I was making some changes.
It is another proff that after a certain amount of time, each error being
repaired leads to at least one other error

I tested it with my sdrplay device, not with an rtlsdr device (yet)

If you encounter any other issue (problem), do not hesitate to contact me

best
jan

andimik, 2019-03-21:
Thanks for the report from South Africa. If it works could you upload an eti file (compress it and attach up to 10 MBytes here).

For cloning an older commit I found the solution on https://stackoverflow.com/questions/7539130/go-to-particular-revision

`git reset`

Chris-v-d-West, 2019-03-21:
Hi Jan,

It works now!  I have been able to pipe to dablin_gtk and play any channel.  When compiling fib-processor.cpp produces a lot of warning.   I see what andimik is saying about cloning and older commit, but could you consider making a release at this point in time, so we know what worked?

Andimik, I will send an eti  in time.  Thanks for your qt-dab and eti-tools!

Thanks again Jan.

Regards,
Chris

JvanKatwijk, 2019-03-21:
The difference between the latest version and previous versions is - apart
from some minor errors - in handling the different devices.
Different devices require different command line parameters (while the
DABsticks just need a gain setting, the SRPlay needs setting
the lnaState and the if gain reduction)
In differentiating for different devices in the "main" program, something
went wrong, which was corrected this afternoon.

I just edited the code for fib-processor such that the warnings in
compiling stay away

andimik, 2019-04-09:
There is an error in the `airspy-handler.cpp`. In line 181 there is no closing )

And in line 536 I get an error 

```
eti-stuff/eti-cmdline/devices/airspy-handler/airspy-handler.cpp:536:40: error: no ‘void airspyHandler::setGain(int)’ member function declared in class ‘airspyHandler’
 void airspyHandler::setGain  (int value) {
                                        ^
CMakeFiles/eti-cmdline-airspy.dir/build.make:518: recipe for target 'CMakeFiles/eti-cmdline-airspy.dir/devices/airspy-handler/airspy-handler.cpp.o' failed
make[2]: *** [CMakeFiles/eti-cmdline-airspy.dir/devices/airspy-handler/airspy-handler.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/eti-cmdline-airspy.dir/all' failed
make[1]: *** [CMakeFiles/eti-cmdline-airspy.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

andimik, 2019-04-09:
It worked in b8f446a ("silly error in sdrplay-handler", 2018-03-24)

JvanKatwijk, 2019-04-09:
fixed (I hope)

closed 2019-05-01 by JvanKatwijk