JvanKatwijk/eti-stuff #43: [eti-cmdline-rtl]: fibquality gets 0 on RPI3B

andimik, 2019-12-09:
Dear Jan, I've lots of PCs and Laptops running with your great tools. But on my Raspberrry Pi 3B (not +) I've discovered that the **eti-cmdline-rtl** does not work properly (no matter if I define a manual gain). **eti-cmdline-rawfiles** works, indeed.

At the start the fibquality is 100 but after a few seconds it gets less and will end at 0. At the same time the CPU load is not much (see screenshot).

[image]

On the same RPI3 qtdab works without any problems, even with Protection level 4A (used in Slovenia).

[image]
[image]

JvanKatwijk, 2019-12-09:
Tomorrow I'll have a look. did not look for a long time to it

andimik, 2019-12-09:
Additionally, I have to mention that the exported eti file is full of errors.

I also tried to pipe rtl_sdr into eti-cmdline-rawfile but this also went wrong.

Seems there is a buffer overflow soon.

andimik, 2019-12-10:
Thanks for the new commit. 

It's really a bit better on my RPI3, now the counter on left botton is faster and the eti-file contains all service IDs (but is still not errorfree). Screenshots taken with XPADxpert, see https://basicmaster.de/ 

Before:

[image]

After:

[image]

In the next days I will get a RPI4 and will test it again.

JvanKatwijk, 2019-12-11:
I optimized some parts, but the "major breakthrough" failed, due to
problems with the compiler chain.
In the past, two, may be three, years ago, the C++ compiler on the  RPI
system supported inlining and using of
neon instructions, but the current compiler chain fails to do so.
The use of the neon instruction would make a difference of probably around
10 %

andimik, 2019-12-13:
Yesterday I got my raspberry 4 and have success with an Airspy. It is error-free now.

So it is really a matter of processing although the CPU load is not even 100%!

BTW: I still have to solve a driver issue with RTL Sticks (they don't work on my Raspberry Pi 4 due to incompatible USB driver).

BTW: Despite of all dependencies mentioned in the `Qt-DAB` Readme, I also had to install `librtlsdr-dev` on Raspbian.

I will also try your latest commit on the RPI 3B.

Thanks again anyway for the optimization.

JvanKatwijk, 2019-12-13:
Hi Andreas

The librtlsdr on Ubuntu requires blacklisting the device, I am surprised
that the default one on Raspbian is OK

Anyway, the optimizations done were actually long overdue, so it was good
to be triggered to have them done

andimik, 2019-12-14:
I close it due to lack of compute power on a Raspberry.

closed 2019-12-14 by andimik