JvanKatwijk/eti-stuff #44: Can't compile on Ubuntu 18.04

jpuigs, 2020-05-02:
```
jordi@jordi-MS-7514:~/Downloads/eti-stuff/eti-cmdline/build$ cmake .. -DRTLSDR=on
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.7
-- Found FFTW3F: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found LibSndFile: /usr/lib/x86_64-linux-gnu/libsndfile.so
-- Found LibSampleRate: /usr/lib/x86_64-linux-gnu/libsndfile.so
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0-25-gd794
-- Found librtlsdr: /usr/local/include, /usr/local/lib/librtlsdr.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jordi/Downloads/eti-stuff/eti-cmdline/build
jordi@jordi-MS-7514:~/Downloads/eti-stuff/eti-cmdline/build$ make
Scanning dependencies of target eti-cmdline-rtlsdr
[  4%] Building CXX object CMakeFiles/eti-cmdline-rtlsdr.dir/main.cpp.o
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp: In function ‘int main(int, char**)’:
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:374:11: error: ‘deviceGain’ was not declared in this scope
           deviceGain = atoi (optarg);
           ^~~~~~~~~~
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:374:11: note: suggested alternative: ‘serviceid’
           deviceGain = atoi (optarg);
           ^~~~~~~~~~
           serviceid
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:380:11: error: ‘autoGain’ was not declared in this scope
           autoGain = true;
           ^~~~~~~~
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:380:11: note: suggested alternative: ‘autogain’
           autoGain = true;
           ^~~~~~~~
           autogain
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:433:39: error: ‘deviceGain’ was not declared in this scope
                                       deviceGain,
                                       ^~~~~~~~~~
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:433:39: note: suggested alternative: ‘serviceid’
                                       deviceGain,
                                       ^~~~~~~~~~
                                       serviceid
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:434:39: error: ‘autoGain’ was not declared in this scope
                                       autoGain,
                                       ^~~~~~~~
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:434:39: note: suggested alternative: ‘autogain’
                                       autoGain,
                                       ^~~~~~~~
                                       autogain
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:435:39: error: ‘deviceIndex’ was not declared in this scope
                                       deviceIndex);
                                       ^~~~~~~~~~~
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:435:39: note: suggested alternative: ‘deviceHandler’
                                       deviceIndex);
                                       ^~~~~~~~~~~
                                       deviceHandler
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:203:10: warning: unused variable ‘gain’ [-Wunused-variable]
 int16_t  gain  = 50;
          ^~~~
/home/jordi/Downloads/eti-stuff/eti-cmdline/main.cpp:204:7: warning: unused variable ‘autogain’ [-Wunused-variable]
 bool  autogain = false;
       ^~~~~~~~
CMakeFiles/eti-cmdline-rtlsdr.dir/build.make:62: recipe for target 'CMakeFiles/eti-cmdline-rtlsdr.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/eti-cmdline-rtlsdr.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/eti-cmdline-rtlsdr.dir/all' failed
make[1]: *** [CMakeFiles/eti-cmdline-rtlsdr.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
jordi@jordi-MS-7514:~/Downloads/eti-stuff/eti-cmdline/build$ 
```

JvanKatwijk, 2020-05-02:
change the main program
elif   HAVE_RTLSDR
int16_t         deviceGain      = 50;
bool            autoGain        = false;
int16_t         ppmOffset       = 0;
int             deviceIndex     = 0;
const char      *optionsString  = "D:d:M:B:C:O:R:G:Qp:";
#elif   HAVE_WAVFILES

jpuigs, 2020-05-02:
OK. Thanks.
There are two warnings of unused variables, but it compiles correctly.

```
/home/jordi/Downloads/eti-stuff/eti-cmdline/src/eti-handling/fic-handler.cpp:
In member function ‘void ficHandler::process_ficInput(int16_t*, int16_t, uint8_t*, bool*)’:
/home/jordi/Downloads/eti-stuff/eti-cmdline/src/eti-handling/fic-handler.cpp:175:9:
warning: unused variable ‘input_counter’ [-Wunused-variable]
 int16_t input_counter = 0;
         ^~~~~~~~~~~~~
/home/jordi/Downloads/eti-stuff/eti-cmdline/src/eti-handling/fic-handler.cpp:176:12:
warning: unused variable ‘k’ [-Wunused-variable]
 int16_t i, k;
            ^
```

closed 2020-06-08 by JvanKatwijk