JvanKatwijk/eti-stuff #51: FR: System Date/Time from Ensemble

richard-gravy, 2021-06-12:
Hello,

I did some reading up after noticing how quickly my portable DAB radio obtains the date and time after power on.
The information is transmitted in local time offset, once per second, in Fig 0/10.
Do you think it might be possible to use this as a time source for an NTP server?

I know that there are better solutions available, like receiving MSF or local equivalent, but the thought occurred to me that if you are receiving DAB on an infrastructure device, like a beefy router, it may be useful for more than just audio.

andimik, 2021-06-13:
It's similar to https://github.com/AlbrechtL/welle.io/issues/654

JvanKatwijk, 2021-06-13:
> It's similar to AlbrechtL/welle.io#654

It is not difficult to put the time on an IP port, such that interested
parties could use it.

JvanKatwijk, 2021-06-13:
Just as a gadget I added the possibility to send the time to an IP port
(8889 by default).
Format (*byes")  FF 00 FF 00 (year & 0xFF) >> 8, year& 0xFF, month, dat,
minutes, seconds
local time

Port can be set by command line parameter -C XXX

configuring by uncommenting datastreamer

as said, it is merely agadget, just to see what was needed to implement
such a feature

richard-gravy, 2021-06-13:
@andimik Pleased that someone else has asked for the same thing in another project, means it's not a terrible idea!

I've just been looking at the "sock" driver for chrony, which is a common NTP daemon on Linux that can be client or server.
In particular, the structure of the data, indicated at [https://git.tuxfamily.org/chrony/chrony.git/tree/refclock_sock.c](https://git.tuxfamily.org/chrony/chrony.git/tree/refclock_sock.c), "struct sock_sample".

Would it be as simple as setting "tv" to the value from the ensemble, set everything else to zero except for the magic number?
Again, it's not practical as a primary reference because you can't interact with the transmission to compensate for drift, but it's better than nothing.