JvanKatwijk/eti-stuff #57: segfault invalid next size

andimik, 2023-01-17:
When trying to convert a raw file from German MDR Saxony multiplex I got from a German user which has a re-config, I got the following seg-fault:

```
$ LANG=C eti-cmdline-rawfiles -F Downloads/9A_MDR\ SACHSEN_2023-01-16_045841.raw > /tmp/test.eti
Period = 8000
program	 ( 1)	 MDR SACHSEN BZ  	 D7C1 is in the list
program	 ( 2)	 MDR SACHSEN SORB	 D7C4 is in the list
program	 ( 3)	 MDRSACHSEN EXTRA	 D9C1 is in the list
program	 ( 4)	 MDR SchlagerW SN	 D6D7 is in the list
program	 ( 5)	 R.SA            	 14C2 is in the list
there might be a DAB signal here
ensemble MDR SACHSEN      detected
program	 ( 6)	 MDR KLASSIK     	 D3D6 is in the list
program	 ( 7)	 MDR KULTUR      	 D3C3 is in the list
program	 ( 8)	 MDR AKTUELL     	 D3D5 is in the list
program	 ( 9)	 MDR JUMP        	 D3C2 is in the list
program	 (10)	 MDR SPUTNIK     	 D3D4 is in the list
program	 (11)	 MDR TWEENS      	 D3D3 is in the list
program	 (12)	 MDR SACHSEN DD  	 D4C1 is in the list
program	 (13)	 MDR SACHSEN L   	 D6C1 is in the list
program	 (14)	 MDR SACHSEN C   	 D5C1 is in the list

yes, here we go
Handling ensemble MDR SACHSEN     until you quit
changes in config not supported, choose again 100
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again
changes in config not supported, choose again 100
free(): invalid next size (normal)
Aborted (core dumped)
```

```
Thread 4995 "eti-cmdline-raw" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe6ffd640 (LWP 45580)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737068914240) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: Datei oder Verzeichnis nicht gefunden.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737068914240) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737068914240) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737068914240, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff7a88476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff7a6e7f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff7acf6f6 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7c21b8c "%s\n")
    at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff7ae6d7c in malloc_printerr (str=str@entry=0x7ffff7c247f8 "free(): invalid next size (normal)") at ./malloc/malloc.c:5664
#7  0x00007ffff7ae8d5c in _int_free (av=0x7fffd8000030, p=0x7fffd8000b60, have_lock=<optimized out>) at ./malloc/malloc.c:4596
#8  0x00007ffff7aeb4d3 in __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3391
#9  0x00007ffff7e512b3 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007ffff7adab43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#11 0x00007ffff7b6ca00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

JvanKatwijk, 2023-01-17:
> #2  __GI___pthread_kill (threadid=140737068914240, ***@***.***=6) at ./nptl/pthread_kill.c:89
> #3  0x00007ffff7a88476 in __GI_raise ***@***.***=6) at ../sysdeps/posix/raise.c:26
> #5  0x00007ffff7acf6f6 in __libc_message ***@***.***=do_abort, ***@***.***=0x7ffff7c21b8c "%s\n")
> #6  0x00007ffff7ae6d7c in malloc_printerr ***@***.***=0x7ffff7c247f8 "free(): invalid next size (normal)") at ./malloc/malloc.c:5664

I'll have a look

JvanKatwijk, 2023-01-18:
Interesting question. In Qt-DAB I solved that a dew years ago, the
implementation is pretty complex, since one has to double
a structure, one "before"  and one "after"  a change configuration.
It will take some time before that particular issue is solved