JvanKatwijk/eti-stuff #59: repeat after EOF for xml broken

andimik, 2023-01-18:
Whereas `eti-cmdline-rawfiles` and `eti-cmdline-wavfiles` play a file from the beginning after reaching EOF, the `eti-cmdline-xml` does not.

```
bool	xml_fileReader::restartReader (int32_t freq) {
	(void)freq;
	if (theReader != nullptr)
	   return true;ifnot
	theReader	= new xml_Reader (theFile,
	                                  theDescriptor,
	                                  5000,
	                                  &_I_Buffer,
	                                  continue_on_eof);
	return true;
}
```

I guess it has to do as you need to skip some bytes.

JvanKatwijk, 2023-01-28:
It took a while, but I had a look.
continuous playing of an xml file is switched off by default. By touching
the continuous button ot worked file.
I changed the layout of the control widget, and had the continuousbutton
indicate whether or not the mode "continuous" is on

andimik, 2023-01-28:
Jan, I was talking about `eti-cmdline-xml`, not Qt-DAB... 😁

JvanKatwijk, 2023-01-29:
I am too busy apparently to look good.
Few remarks on xml files in eti
The text in the main program clearly states
#elif   HAVE_XML_FILES
#include        "xml-filereader.h"      // does not work yet
#endif

However, your mods seem to have worked.
There was one thing in the code that was missing:
Saving the "continue_on_eof" flag from the xml-filereader, i.e. the
interfacehandler, so an uninitialized "continue_on"eof" was
passed on to xml-reader, i.e. the worker
So, after adding a line "this -> continue_on_eof = continue_on_eof" fixed
that

There is one issue though and I'll try to handle that one of these days and
- the same for Qt-DAB btw - is that at the end of reading the xml file
some garbage is put into the buffer. Fxing that requires more than a single
line of code

andimik, 2023-01-29:
Fixed in [60eb2d2](https://github.com/JvanKatwijk/eti-stuff/commit/60eb2d286b6ec181094d78c9b399529fb5640ff2), thanks!

closed 2023-01-29 by andimik