JvanKatwijk/eti-stuff #62: eti-cmdline-rtlsdr does not find DAB+ channels, dab-scanner-rtlsdr shows

arkadirenko, 2024-03-30:
I wish to use eti-cmdline-rtlsdr in conjunction with dablin, to listen to DAB+ channels.
I did a search for avalilable channels with dab-scanner-rtlsdr with the following results:
```
Ensemble SWR BW N        ; ensembleId 10EA; channel 9D; 

Audio services
type;program name;channel;serviceId;subchannelId;start address;length (CU); bit rate;DAB/DAB+; genre; prot level; code rate; language

audioservice;SWR4 MA         ;9D;DC04;10;570;84;112;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;SWR4 HN         ;9D;DD04;8;402;84;112;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;SWR4 S          ;9D;D904;5;318;84;112;DAB+;0;EEP 3-A;1/2;unknown;
audioservice;SWR4 KA         ;9D;D804;9;486;84;112;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;SWR Aktuell     ;9D;D3A6;1;0;54;72;DAB+;3;EEP 3-A;1/2;unknown;
audioservice;SWR2            ;9D;D3A2;3;138;96;128;DAB+;7;EEP 3-A;1/2;unknown;
audioservice;SWR1 BW         ;9D;D301;2;54;84;112;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;DASDING         ;9D;D3A5;13;738;84;112;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;SWR3            ;9D;D3A3;4;234;84;112;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;SWR4 UL         ;9D;DE04;12;654;84;112;DAB+;9;EEP 3-A;1/2;unknown;

Data Services
type;program name;channel;serviceId;subchannelId;start address;length (CU); bit rate; FEC; prot level; appType; DSCTy; subService; 

dataservice;EPG              (data);9D;E0D010C9;20;822;18;24;no FEC;EEP 3-A;SPI, was EPG;Multimedia Object Transfer (MOT);no;
dataservice;TPEG             (data);9D;E0D020EB;21;840;12;24;no FEC;EEP 4-A;TPEG;Transparent Data Channel (TDC);no;

Ensemble SWR RP          ; ensembleId 10D9; channel 11A; 

Audio services
type;program name;channel;serviceId;subchannelId;start address;length (CU); bit rate;DAB/DAB+; genre; prot level; code rate; language

audioservice;SWR1 RP         ;11A;D3A1;2;108;72;96;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;SWR Aktuell     ;11A;D3A6;1;54;54;72;DAB+;3;EEP 3-A;1/2;unknown;
audioservice;SWR4 LU         ;11A;D8A4;9;594;66;88;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;SWR4 TR         ;11A;DCA4;10;660;66;88;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;DASDING         ;11A;D3A5;5;324;72;96;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;SWR3            ;11A;D3A3;4;252;72;96;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;SWR4 MZ         ;11A;DAA4;6;396;66;88;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;SWR4 KO         ;11A;D6A4;8;528;66;88;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;SWR4 KL         ;11A;D7A4;7;462;66;88;DAB+;9;EEP 3-A;1/2;unknown;
audioservice;RPR1.           ;11A;D3A8;0;0;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;bigFM           ;11A;100F;11;726;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;Rockland Radio  ;11A;D3AA;12;780;54;72;DAB+;11;EEP 3-A;1/2;unknown;
audioservice;SWR2            ;11A;D3A2;3;180;72;96;DAB+;7;EEP 3-A;1/2;unknown;

Data Services
type;program name;channel;serviceId;subchannelId;start address;length (CU); bit rate; FEC; prot level; appType; DSCTy; subService; 

dataservice;EPG              (data);11A;E0D010D9;15;834;18;24;FEC;EEP 3-A;SPI, was EPG;Multimedia Object Transfer (MOT);no;
dataservice;TPEG             (data);11A;E0D020EB;16;852;12;24;no FEC;EEP 4-A;TPEG;Transparent Data Channel (TDC);no;

Ensemble OAS BW          ; ensembleId 10C9; channel 11B; 

Audio services
type;program name;channel;serviceId;subchannelId;start address;length (CU); bit rate;DAB/DAB+; genre; prot level; code rate; language

audioservice;Schlager Radio  ;11B;1735;17;756;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;HITRADIO OHR    ;11B;D40D;8;324;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;RADIO REGENBOGEN;11B;D308;9;378;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;DONAU 3 FM      ;11B;DA08;7;270;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;REGENBOGEN 2    ;11B;D30E;10;432;54;72;DAB+;11;EEP 3-A;1/2;unknown;
audioservice;Radio Seefunk   ;11B;140A;12;540;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;RADIO TEDDY     ;11B;1B2E;14;648;54;72;DAB+;18;EEP 3-A;1/2;unknown;
audioservice;ANTENNA BW      ;11B;130B;18;810;48;64;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;baden.fm        ;11B;1702;4;108;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;DIE NEUE 107.7  ;11B;130A;5;162;54;72;DAB+;11;EEP 3-A;1/2;unknown;
audioservice;Radio Ton       ;11B;D70D;11;486;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;die neue welle  ;11B;D50C;6;216;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;antenne1        ;11B;D30A;15;702;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;bigFM           ;11B;100F;13;594;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;egoFM           ;11B;1014;3;54;54;72;DAB+;15;EEP 3-A;1/2;unknown;
audioservice;RADIO 7         ;11B;D30B;1;0;54;72;DAB+;10;EEP 3-A;1/2;unknown;
```

when I try now to play some of those channels with dablin I use this command line for Channel 9B, 11A, 11B
**_./eti-cmdline-rtlsdr -C 9B -G 60 | dablin_gtk_**
and no matter, which gain mode (Q or 0-60) I use, the result is:
```
Found Fitipower FC0012 tuner
SDLOutput: using SDL version '2.26.5'
EnsembleSource: reading ETI from 'stdin'
samplerate set to 2048000
Supported gain values (5): -9.-9 -4.0 7.1 17.9 19.2 
effective gain: gain 17.9
There does not seem to be a DAB signal here
EnsembleSource: EOF reached!
```
what I am doing wrong?

andimik, 2024-03-31:
@arkadirenko 

I have (among others of course) a similar (but not the same old) stick at home (mine is Noxon DAB stick v2) with the same tuner manufacturer but have no problems with `eti-cmdline-rtlsdr` at all.

But it's strange your stick seems to support only a few gain settings.

What were the steps when you compiled `eti-cmdline`?

Don't you have another stick?

```
$ eti-cmdline-rtlsdr -Q -C 6A | dablin_gtk 
tunedFrequency =  181936000
DABlin v1.14.0-4-g9a99ed9 - capital DAB experience
Plays a DAB/DAB+ audio service from an ETI-NI or an EDI AF stream.
  https://github.com/Opendigitalradio/dablin

Found Fitipower FC0013 tuner
SDLOutput: using SDL version '2.0.20'
EnsembleSource: reading ETI from 'stdin'
samplerate set to 2048000
Supported gain values (27): 0.0 3.1 6.5 10.3 13.9 17.7 21.7 25.7 29.3 33.3 37.3 40.1 43.9 47.9 51.9 55.9 59.9 63.9 67.9 71.9 75.9 79.9 83.9 87.9 91.9 95.9 99.9 
effective gain: gain 47.9
Allocating 15 zero-copy buffers
there might be a DAB signal here
ensemble DAB+ Austria     detected
program	 ( 1)	 Radio Maria     	 A3DD is in the list
program	 ( 2)	 ENERGY          	 AC51 is in the list
program	 ( 3)	 KLASSIK RADIO   	 AD53 is in the list
program	 ( 4)	 arabella RELAX  	 AD54 is in the list
program	 ( 5)	 j�.live         	 AD56 is in the list
program	 ( 6)	 Rock Antenne    	 AD27 is in the list
program	 ( 7)	 #Radio ONE      	 AD2A is in the list
program	 ( 8)	 ERF S�d         	 AD24 is in the list
program	 ( 9)	 Mein Kinderradio	 AD55 is in the list
program	 (10)	 *STEPHANSDOM*   	 AC52 is in the list
program	 (11)	 'WELLE 1        	 AD58 is in the list
program	 (12)	 arabella HOT    	 AD59 is in the list
program	 (13)	 Antenne �sterrei	 AD57 is in the list
program	 (14)	 Radio Flamingo  	 AD5A is in the list
program	 (15)	 -AUSTRIA        	 A3E0 is in the list
program	 (16)	 * 88.6 *        	 AC47 is in the list

yes, here we go
```

Also with E4000 Tuner it works:

```
$ eti-cmdline-rtlsdr -Q -C 6A | dablin_gtk 
tunedFrequency =  181936000
DABlin v1.14.0-4-g9a99ed9 - capital DAB experience
Plays a DAB/DAB+ audio service from an ETI-NI or an EDI AF stream.
  https://github.com/Opendigitalradio/dablin

Found Elonics E4000 tuner
SDLOutput: using SDL version '2.0.20'
EnsembleSource: reading ETI from 'stdin'
samplerate set to 2048000
Supported gain values (20): 0.0 2.9 6.0 8.9 11.9 14.7 17.6 20.6 23.5 26.4 29.4 32.3 35.3 38.2 40.8 43.6 46.6 49.5 52.1 54.8 
effective gain: gain 29.4
Allocating 15 zero-copy buffers
program	 ( 1)	 Mein Kinderradio	 AD55 is in the list
program	 ( 2)	 *STEPHANSDOM*   	 AC52 is in the list
program	 ( 3)	 'WELLE 1        	 AD58 is in the list
program	 ( 4)	 arabella HOT    	 AD59 is in the list
program	 ( 5)	 Antenne �sterrei	 AD57 is in the list
program	 ( 6)	 Radio Flamingo  	 AD5A is in the list
program	 ( 7)	 -AUSTRIA        	 A3E0 is in the list
program	 ( 8)	 * 88.6 *        	 AC47 is in the list
there might be a DAB signal here
ensemble DAB+ Austria     detected
program	 ( 9)	 Radio Maria     	 A3DD is in the list
program	 (10)	 ENERGY          	 AC51 is in the list
program	 (11)	 KLASSIK RADIO   	 AD53 is in the list
program	 (12)	 arabella RELAX  	 AD54 is in the list
program	 (13)	 j�.live         	 AD56 is in the list
program	 (14)	 Rock Antenne    	 AD27 is in the list
program	 (15)	 #Radio ONE      	 AD2A is in the list
program	 (16)	 ERF S�d         	 AD24 is in the list

yes, here we go
```

arkadirenko, 2024-03-31:
@andimik 
no, currently this is the only stick I have. I compiled eti-cmdline just in the same environment, in the same bash session, as dab-scanner-rtlsdr .
I cloned both projects from github, created a build directory below, started `cmake ../ -DRTLSDR=ON && make`in the build directory. That's all.
The antenna is fixed on my window and has not changed between all my tries.
When i go back an do a scan, i get the same results.

But its a good hint, that your stick has much more and perhaps higher gain possibilities! Without a comparision, this was just a information.

andimik, 2024-03-31:
@arkadirenko 

Maybe you post the console output of `dab-scanner` like mine:

```
$ ./linux-bin/dab-scanner-1.8 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
main: Detected system language "de_DE"
main: Set language "de_DE"
OK, functions seem to be loaded
FIR Filter 2400 kHz
Found Fitipower FC0013 tuner
Supported gain values (19): 70,3 66,3 62,3 58,3 54,3 50,3 46,3 42,3 38,3 34,3 30,3 26,3 22,3 18,3 14,3 10,3 6,5 3,1 0,0 
NOXON DAB Stick 0
dirName = /home/andreas
suggested filename = /home/andreas./dab-scanner-So.-März-31-2024-16-11-55.txt
Allocating 15 zero-copy buffers
```

It won't tell you the used gain setting, but it should display the same possible gain values.

arkadirenko, 2024-03-31:
@andimik 
my dab-scanner tells me also the used gain and the same 5 possibilities:
```
dab_scanner V 1.0alfa,
Copyright 2018 J van Katwijk, Lazy Chair Computing
2018 Hayati Ayguen
2019 J van Katwijk
2020 J van Katwijk
going for rtlsdr 174928000 800
OK, functions seem to be loaded
Found Fitipower FC0012 tuner
samplerate set to 2048000
Supported gain values (5): -9.-9 -4.0 7.1 17.9 19.2 
effective gain: gain 19.2
checking data in channel 5A
dab processor will stop
```
a google search to `"Fitipower FC0013" "Supported gain values"` shows me, that there are variations between only four up to 29, with the majority of 23 gain values.

but much more important: my dab-scanner tells me, that I use **V 1.0alfa** while you are using **version 1.8** 

so I realized, that mine is from https://github.com/JvanKatwijk/dab-cmdline/tree/master/dab-scanner, while yours is a standalone projekt.

I am now in the process building this as well.  But: the scanner isn't the problem, as also the alfa version reports me all local channels

andimik, 2024-03-31:
Could you try this fork: https://github.com/hayguen/eti-stuff ? It's a bit behind Jan's version.

Because we are using this fork in the FMLIST scanner for RTLSDR sticks, where it must work.

JvanKatwijk, 2024-04-02:
Apology for the late reply,
The first thing I nore is that the gain you specify is not supported
The gain for sticks is relative - since different sticks support difefrent
gain levels,
so 100 is the maximum gain

The second observation is that - depending on the signals strength, I do
not know what kind of antenna you ar eusing -
it takes more time than the period set by default to answer the question
whether or not a DAB signl is in the air

With options D and d you can set the delay, i.e. the time after which - if
no signal is found - the software gives up, and
the time the software is allowed to detect some services.

An alternative to investigate your situation is to look at dab-cmdline,
which does the whole :  but with a preset service in a preset channel

arkadirenko, 2024-04-03:
did some more testing. Dab-cmdline is giving me the same results as dab-scanner-rtlsdr.
As of the gain I realized by testing, that a G value of zero gives me the lowest gain, a value of 99 the highest gain, while 100 gives me an effective gain of 0, which is not in the list of the 5 supported gain values.
The older version, linked from @andimik just behaves the same.
I got hold of another stick, but it turned out, that this is identical to mine in responses and behaviour.

Looking again at the output, I wonder about the lowest gain:
_Supported gain values (5): **-9.-9** -4.0 7.1 17.9 19.2_ 
How it happens that the fractional portion is also negative?
In most examples I found in the internet, the lowest gain is **_-9.9_**, like in this example: 
_Supported gain values (23): **-9.9** -7.3 -6.5 -6.3 -6.0 -5.8 -5.4 5.8 6.1 6.3 6.5 6.7 6.8 7.0 7.1 17.9 18.1 18.2 18.4 18.6 18.8 19.1 19.7_

In other examples gain values are starting with 0 and going up to 49.9:
_Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 128.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6_

I try now to get another stick with another tuner, maybe I have more luck with this.

JvanKatwijk, 2024-04-03:
what kind of antenna you ar eusing, and do you have aby idea what the
distance is to the neaeest transmitter?
With my Noxon (pretty old) and my whip antenna - sitting in the living room
ground floor - I do not get a strong enough signal,
however, if I go up to the second floor I do get one or two ensembles here.
Yhe transmitetrs here are within a distance
of app 10 to 15 Km

arkadirenko, 2024-04-03:
I use a small antenna, where i mounted a metall plate as groundplane.
For all my tests I glue the antenna stick on the window in 3rd floor with free view to the horizon, no neighbour buildings.
With the antenna on my table (2m from the window) receiving is worse, but still possible.
[image]

JvanKatwijk, 2024-04-04:
But do you get any signal?
The problem is of course that neither the eit nor the dab-cmdline software
visualizes the signal.
If you do not get results with the command line tools I suggest you use
Qt-DAB (or whatever other program)
that helps you to see more of the incoming signal.
For Linux type systems there is an appImage, for Windows systems an
installer

andimik, 2024-04-04:
@arkadirenko 

What about `eti-cmdline-rawfiles`? Can you test it with a (good) prerecorded file?

Want to know if the culprit is the signal, the stick (gain) or `eti-cmdline-rtlsdr`

arkadirenko, 2024-04-06:
I played a little bit more around and played with welle.io.  After using welle.io the stick worked perfect together with eti-cmdline!
After replugin the stick I had again no output, using welle.io before my next try, the stick worked.  Now it worked also after several replugs. So I am clueless and think, that this cheap stick might have some glitches.  The odd announcement of gainvalue "-9.-9" is also not capable rising my trust. So I ordered meanwhile an original rtl-sdr stick, for further investigation.

Meanwhile you could give me possibly some more information, about the channel information, dab-scanner reports.

Comparing some scans I did in the last days, I figured out, that the value of subchannel-Id where different for about half of all programs:
```audioservice;bigFM           ;11A;100F;11;726;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;bigFM           ;11A;100F;14;780;54;72;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;DASDING         ;11A;D3A5;5;270;72;96;DAB+;10;EEP 3-A;1/2;unknown;
audioservice;DASDING         ;11A;D3A5;5;324;72;96;DAB+;10;EEP 3-A;1/2;unknown; 
```
What is the purpose of Subchannel-Id and service-Id?  Can you recommend me some beginners documentation about the dab protocol?

JvanKatwijk, 2024-04-08:
serviceIds are the indernal representations of the service, so each
servicename is bound to a single serviceId.
De subchannelId identifies a description of the processing of a certain
segment of data to output, either as audio or data.
While not happening often, a serviceId is the name for more than a single
outut. The main output is then most often audio, but the
"sub" service(s) mostly contain data. In Germany we see services contain
audio and journaline (the latter a kind of teletext)
Some years ago, the Belgian transmissions had services with the audio as
"main" service and the slide shows as subservice(s)

arkadirenko, 2024-04-08:
@JvanKatwijk

hmm, that explanation sounds logical. However my observations are not matching.  If the subchannel-id would differentiate for example audio and text-information, I would expect, that I see per service at least sometimes two lines with both sub-channel-ids.
However I did today 30 scans with a delay of 10min in between and found only one subchannel-id per service.  In my older files I see the different subchannels only at different days. Perhaps there was a kind of change in the eastertime? I will continue with some tests over the next days.

arkadirenko, 2024-04-12:
I received today my new original rtl-sdr v4.
I can now receive almost two times more senders, have a wide gain range, so there is really a big difference.
The command line version of dab-scanner-rtl responds:
```
going for rtlsdr 174928000 99
OK, functions seem to be loaded
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
samplerate set to 2048000
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
```
And the promised backwards connectivity of the v4 drivers (https://www.rtl-sdr.com/v4/) seems to work.
So I assume, that my initial problems are related to the cheap stick. I will close this issue  now.
I have learned a lot, thanks for all your help @JvanKatwijk  and @andimik !

closed 2024-04-12 by arkadirenko

andimik, 2024-04-12:
Oh, even much less CPU power! Wow!

JvanKatwijk, 2024-04-13:
Apologies for not responding, I have/had this week the flu and not
functioning optimally
Wrt the observation of subchannel. The subchannelId is mrely a number
between o and 64. The subchannel itself is a descriptor with address and
encoding information
on the segment of samples in the decoded data that it described.
Common practice in (most) DAB transmittions is to have a single SId with a
single component Id and a single subchannelId, however, at leasrt some
german transmissions
carry as subservice journaline data. So, if you would have a look at the
content table (or scan result) of my Qt-DAB software you would see the same
service being mentioned
twince, once in the part for audio descriptions, one for the data
descriptions.

Anyway, glad that you have a V4 stick with good results
best
jan