JvanKatwijk/eti-stuff #64: ability to view TII from a DAB transmitter

jeremyp3, 2025-06-18:
I'd like to be able to see the TII of a transmitter. It's very useful in the case of dab propagation to know which transmitter you're hanging on to when there are several on the same channel in SFN.

does this possibility exist, and i haven't found it?

thanks !

jeremyp3, 2025-06-18:
as long as it stays in cli, that's fine with me :) thanks ! 
I'll wait then before opening a issue on the aforementioned eticmdline repo

JvanKatwijk, 2025-06-22:
> It is just the dab-cmdline with soe (well, quite a few) additions

I made a first modification of example-2 in the dab-cmdline repo that adds
a first version of TII decoding
as a sample of the output reads:

16 1 -> DE 5C DR Deutschland Hamburg/Heinrich-Hertz-Turm 53.563122 9.975833

16 2 -> DE 5C DR Deutschland Hamburg/Moorfleet 53.519230 10.102853

telling the mainId, subId, country, channel, ensembleName, transmitter name
and transmitter coordinates

It is Work In Progress, but you might want to take a look

jeremyp3, 2025-06-23:
after adding the en_US.UTF-8 locale, i managed to run the program.

after a quick test on the transmitters near me, it seems to work. even if i don't have the database, i can see the 2 tii numbers, (mainId and subId) which then allows me to search in a csv. i'm waiting to get some dab propagation to see how it behaves in this case, but for me it's a very good start! :)

thanks!

JvanKatwijk, 2025-06-23:
The "how to use" was not updated yet.
You need a dabase, that can be downloaded from the Qt-DAB reposiitory
The database should be named ".txdata.tii" and placed in the home dorectory

Btw it seems more logical to create an issue in dab-cmdline

(to be continud later on)

JvanKatwijk, 2025-06-24:
I started an issue on dab-cmdline, feel free to comment there